irfu/irfu-matlab #16: Use the new stable nasa_cdf_patch, prefix 'SPDF', in all cdf functions.

thomas-nilsson-irfu, 2015-04-28:
Hi,
NASA have just (minutes ago, after hopefully my last bug report) released the new stable cdf patch for Matlab. (v.3.6). This is now able to handle the upcoming leap second of 2015/07/30 while ensuring backwards/forwards compatibility with any future leap seconds by adding a field in each cdf file with information about the last leap second used when writing the file.

One major change from previous version is that: "All SPDF-distributed modules are now renamed with a prefix of 'SPDF', e.g., SPDFCDFREAD replaces CDFREAD, SPDFCDFWRITE replaces CDFWRITE, etc. Each module still keeps its original functionality and may have a few new enhancements/features."

All MMS specific code has already been modified to reflect this behavior but as there are still a lot of places in irfu-matlab where the old functions are used I am creating this issue. When all instances of cdfread, cdfwrite, cdfinfo etc. are replaced this issue can be closed and the new stable cdf-patch can fully replace the nasa_cdf_patch.

Please feel free to help locate and replace the obsolete function calls.

// Thomas

vaivads, 2015-04-29:
Hi Thomas,

great news!

This also avoids confusion with matlab cdfread as both can have different
syntax.
It is most important to update and test @dataobj because I think it is used
also by MMS
and currently there is still bare "cdfread".

Andris

thomas-nilsson-irfu, 2015-04-29:
One bug was discovered in the version uploaded yesterday. It assumes LD_LIBRARY_PATH (linux) or PATH (windows) has been configured on the system to include the directory with libcdf.so (or the dll for Windows OS). I added some code to irf.m in MMSdevel which should ensure the path variable is configured accordingly if it is missing.

But I do not run Mac OS and am unsure which parameter is used for Mac. Someone running Mac OS, please add the corresponding environment variable so that this commit can move to master.

closed 2015-06-11 by thomas-nilsson-irfu