irfu/irfu-matlab #17: 'spdfcomputett2000c' undefined

binlipric, 2015-05-07:
I am using irfu-matlab version:  v1.8.0
The issue is as following: 

>> irf
irfu-matlab version:  v1.8.0
Checking if you have latest irfu-matlab... YES:)
loading datastore...
IRBEM is OK
CDF_LEAPSECONDSTABLE was not set in user environment.
Automatically setting it to: C:\Users\lenovo\Documents\MATLAB\irfu-matlab-master\contrib\nasa_cdf_patch\CDFLeapSeconds.txt
>> Example_1
Moving to temporary directory: C:\Users\lenovo\AppData\Local\Temp\tp53ddfefa_0535_44b7_974f_32171da6011b
Undefined function 'spdfcomputett2000c' for input arguments of type 'double'.

Error in spdfcomputett2000 (line 57)
out = spdfcomputett2000c (datetime);

Error in irf_time (line 115)
		t_out = spdfcomputett2000(t_in);

Error in irf_time (line 82)
		t_temp=irf_time(t_in,[format_in '>ttns']);

Error in Example_1 (line 12)
tint=[irf_time([2006 9 27 17 17 0]) irf_time([2006 9 27 17 24 0])]; % time interval

thomas-nilsson-irfu, 2015-05-08:
This is rather strange, the mex compiled versions of "spdfcomputett2000c" (for Linux, Mac and Windows) are included in the same path as "spdfcomputett2000". So when you're able to call on function spdfcomputett2000 (in irf_time) this function should be able to call on spdfcomputett2000c.

Could you please reply with the result of the Matlab command "computer".

binlipric, 2015-05-11:
>> computer

ans =

PCWIN

binlipric, 2015-05-11:
OK, I think I found the problem, my computer system is win7 32bit, 'spdfbreakdowntt2000c.mexw64' is for 64bit system, is there some way to fix this??? Thanks a lot!

thomas-nilsson-irfu, 2015-05-11:
This issue should now be solved with v1.8.1... (Hopefully) However as was written in the commit message a .dll file exist in the NASA cdf patch which might be needed by the compiled mex files. This .dll file however has the exact same name in Windows 64 bit patch as in Windows 32 bit patch but its sha512 sum is different. We will not replace this 64 bit dll with the 32 bit version in contrib/nasa_cdf_patch.

If you experience any issue with this on the Windows 32 bit OS, please feel free to contact us again. (One possible solution would then be to add the NASA cdf patch somewhere outside of irfu-matlab and manually add that path in your instance of Matlab).

With best regards,

Thomas Nilsson

closed 2015-05-11 by thomas-nilsson-irfu

binlipric, 2015-05-12:
Thanks a lot! May be change to a 64Bit OS is the best soluation.
I have updated to v1.8.1, the issue is remain,and it's like this : 

>> irf
irfu-matlab version:  v1.8.1
Checking if you have latest irfu-matlab... YES:)
loading datastore...
IRBEM is OK
CDF_LEAPSECONDSTABLE was not set in user environment.
Automatically setting it to: C:\Users\lenovo\Documents\MATLAB\irfu-matlab-master\contrib\nasa_cdf_patch\CDFLeapSeconds.txt
Operating system, Windows 32 bit, is not fully tested with IRFU-MATLAB.
Recommended OS are: Linux, Mac and Windows (all 64 bit).
>> caa_download('list')
irfu-matlab version:  v1.8.1
Checking if you have latest irfu-matlab... YES:)
  [warning: caa_download(454)] Patience! Requesting "list" http://csa.esac.esa.int/csa/aio/metadata-action?&NON_BROWSER&SELECTED_FIELDS=DATASET.DATASET_ID,DATASET.START_DATE,DATASET.END_DATE,DATASET.TITLE&RESOURCE_CLASS=DATASET&RETURN_TYPE=CSV&QUERY=DATASET.DATASET_ID%20like%20''

ans =

     []

>> computer

ans =

PCWIN

>> Example_1
Moving to temporary directory: C:\Users\lenovo\AppData\Local\Temp\tpdf1d2a12_4448_4df9_8f9c_ae15058d79b6
Invalid MEX-file
'C:\Users\lenovo\Documents\MATLAB\irfu-matlab-master\contrib\nasa_cdf_patch\spdfcomputett2000c.mexw32':
C:\Users\lenovo\Documents\MATLAB\irfu-matlab-master\contrib\nasa_cdf_patch\spdfcomputett2000c.mexw32 不是有效的
Win32 应用程序。%%%%(in Chinese,means is not a valid Win32 appliacation---Bin Li add)

Error in spdfcomputett2000 (line 57)
out = spdfcomputett2000c (datetime);

Error in irf_time (line 117)
		t_out = spdfcomputett2000(t_in);

Error in irf_time (line 84)
		t_temp=irf_time(t_in,[format_in '>ttns']);

Error in Example_1 (line 12)
tint=[irf_time([2006 9 27 17 17 0]) irf_time([2006 9 27 17 24 0])]; % time interval

Best regards,
Bin

thomas-nilsson-irfu, 2015-05-12:
This is unfortunate.. 

Other than changing operating system I could only suggest you try to download the NASA cdf patch (for 32 bit Windows) manually and store it somewhere outside of irfu-matlab then and add this separate folder to your Matlab path. (To be on the safe side also comment out/remove line 108 in irf.m, which reads "['contrib' filesep 'nasa_cdf_patch']" so that these files do not interfere with your separate cdf patch).

You can find the NASA cdf patch here along with the instructions for Windows 32 bit:
http://cdf.gsfc.nasa.gov/html/matlab_cdf_patch.html

Should you be unable to get this cdf patch working while only running the separate path there is an e-mail for further cdf support listed on the bottom of that page.

With best regards,

Thomas Nilsson