irfu/irfu-matlab #18: epoch16 bug

ajohlander, 2015-05-28:
Time interval:

tStart = irf_time([2002 02 03 05 19 12]); %start time
tint = [tStart, tStart+12]; %time interval

local.c_read('E_Vec_xyz_ISR2__C3_CP_EFW_L2_E3D_INERT',tint) does not read 

data is stored locally

ajohlander, 2015-05-29:
Now it reads data but the time stamps are incorrect. Time increases in steps of 1 second with ~25 data points with the same time value.

thomas-nilsson-irfu, 2015-06-02:
Hi,
it appears to be a bug in the NASA supplied "spdfbreakdownepoch16c", which is used by irf_time when converting from "cdfepoch16" to "epoch". (called on by line 243 in +local/c_read.m). I have just filed a bug report with Michael and the rest of the cdf team so hopefully they will soon release a new compiled cdf patch.

ajohlander, 2015-06-09:
Now it works. Thanks!

commit 4156b2f7 (2015-06-09): New CDF patch from NASA, v. 3.6.0.4 (dated 2015/06/08). This include a proper conversion to/from cdfepoch16. Fixes #18.
closed 2015-06-09 by ajohlander