irfu/irfu-matlab #20: PEACE density from PITCH_SPIN using s/c potential correction

sergiotr, 2015-06-11:
With the last changes, the following code from irfnotes stopped working. I have been trying to see the problem but no success so far. Maybe some changes in c_caa_var_get ?

if 0, % PEACE calculate density nPEACE1..nPEACE4 [cc] from PITCH_SPIN_DPFlux products using s/c potential correction from EFW
  for ic=1:4,
    c_eval('caa_load C?_CP_PEA_PITCH_SPIN_DPFlux',ic); % to speed up later
    varname=irf_ssub('Spacecraft_potential__C?_CP_EFW_L3_P',ic);
    scpotmat=c_caa_var_get(varname,'mat');
    scpotmat(isnan(scpotmat(:,2)),:)=[]; % remove NaN densities
    varname=irf_ssub('Data__C?_CP_PEA_PITCH_SPIN_DPFlux',ic);
    [var,~,varmat]=c_caa_var_get(varname);
    scpot=irf_resamp(scpotmat,varmat); % interpolate sc potential to PEACE data points
    phi=c_caa_var_get(var.DEPEND_1);
    x=getfield(c_caa_var_get(phi.DELTA_PLUS),'data');phi_dplus=x(1,:);
    x=getfield(c_caa_var_get(phi.DELTA_MINUS),'data');phi_dminus=x(1,:);
    en=c_caa_var_get(var.DEPEND_2);
    x=getfield(c_caa_var_get(en.DELTA_PLUS),'data');en_dplus=x(1,:);
    x=getfield(c_caa_var_get(en.DELTA_MINUS),'data');en_dminus=x(1,:);
    PEACE_energy_channels=en.data(1,:)+0.5*(en_dplus-en_dminus);
    PEACE_phi_min=phi.data(1,:)-phi_dminus;
    PEACE_phi_max=phi.data(1,:)+phi_dplus;
    phi_factor=repmat((cos(PEACE_phi_min*pi/180)-cos(PEACE_phi_max*pi/180))',1,length(PEACE_energy_channels));
    
    nPEACE=[varmat.t(:) varmat.t(:)*0];
    varmat.data(isnan(varmat.data))=0;
    for jj=1:size(nPEACE,1),
      satpot=-scpot(jj,2)*1.3; % assumes that probe t spacecraft potential is ~75% of spacecraft potential
      ii_energy=find(PEACE_energy_channels>satpot); % use only these energy chanels
      [en_min,ii_energy_min]=min(PEACE_energy_channels(ii_energy));
      ii_energy(ii_energy==ii_energy_min)=[]; % remove first channel after satellite potential
      [en_min,ii_energy_min]=min(PEACE_energy_channels(ii_energy));
      ii_energy(ii_energy==ii_energy_min)=[]; % remove second channel after satellite potential
      en_factor=repmat(1e-3*(en_dplus(ii_energy)+en_dminus(ii_energy)).*sqrt(1e-3*(PEACE_energy_channels(ii_energy)-satpot))./(1e-3*PEACE_energy_channels(ii_energy)),length(phi_dplus),1);
      ncoef=ones(length(phi_dplus),length(ii_energy));
      ncoef=ncoef*0.2284e-7*sqrt(1/1836)*2*pi.*phi_factor(:,ii_energy).*en_factor;
      nPEACE(jj,2)=sum(sum(shiftdim(varmat.data(jj,:,ii_energy)).*ncoef));
    end
    c_eval('nPEACE?=nPEACE;',ic);
  end
end

sergiotr, 2015-06-11:
So it seems that the problem is when retrieving data:

x=getfield(c_caa_var_get(phi.DELTA_PLUS),'data');

in irfu_matlab v.1.7.3 x is a 1x12 matrix
in irfu_matlab v.1.8.2 x is a 12x1 matrix

closed 2015-06-11 by sergiotr

thomas-nilsson-irfu, 2015-06-11:
Looks like this previous fix related to issue #18  (currently in the devel & MMSdevel branches) should solve the underlining issue:

https://github.com/irfu/irfu-matlab/commit/1ef206698e51b335a37cb0dee29746383f5d257f