irfu/irfu-matlab #22: Method Tseries.cross and automatically resampling

sergiotr, 2015-08-27:
The last line of code worked in v1.9.4 and stopped working in v1.9.5
tint  = irf.tint('2015-08-15T12:00:05Z/2015-08-15T15:00:00Z'); 
B3=mms.db_get_ts('mms3_dfg_srvy_ql','mms3_dfg_srvy_dmpa',tint);
E3=mms.db_get_ts('mms3_edp_fast_ql_dce2d','mms3_edp_dce_xyz_dsl',tint);
ExB= B3.cross(E3).data

In version v1.9.4 it gave a warning:
[warning: TSeries.cross(420)] cross(T1,T2): resampling T2 TSeries

In v1.9.5 gives error:
Error using TSeries/cross (line 438)
Input TS objects have different timelines, use resample()

Error in TSeries/subsref (line 290)
          [varargout{1:nargout}] = builtin('subsref',obj,idx);

thomas-nilsson-irfu, 2015-08-27:
Hi, this is related to the fact that the time of measurement between instruments does not agree.

There was an implicit resample() done before commit (ce174e90040f5688e26b62a2b68bbd5a6d072f54), which removed this assumption. I believe Yuri intended for it it to be up to you to decide if "E3" should be resampled to align with "B3"s time or if it should be done the other way around.

sergiotr, 2015-08-28:
Thanks for the reply Thomas. I liked the promiscuous way!

vaivads, 2015-08-28:
To avoid ambiguity, e.g.
ExB1= cross(B3,E3.resample(B3))
ExB2= cross(B3.resample(E3),E3)
are different time series, it is better that user
explicitly tells which resampling he wants to make.
Because in practice there can be situations where
sampling can differ a lot and thus the implicit calculations
can lead to very different results depending on the order
of arguments.

Note, there is also a function irf_e_vxb.

sergiotr, 2015-08-28:
Thanks Andris, now I understand better. In fact this use of objects is very powerful, did not thought about simply including resample as you show!

closed 2015-08-28 by sergiotr