irfu/irfu-matlab #23: irf_get_data fails to load entire data object

LoveAlm, 2015-11-30:
Using the latest version, both Master and Devel-MMS, I encounter errors when loading CIS ion data from any dcf-file. If I for example run:

idpf=irf_get_data('Differential_Particle_Flux__C1_CP_CIS-HIA_PAD_HS_MAG_IONS_PF','caa','mat')
It fails to load the data on the width of the energy bins, i.e. the variables idpf.dep_x{2,1}.df.minus and idpf.dep_x{2,1}.df.plus. Instead of loading the full vector it loads a single value from the vector. Likely the first value. The rest of the variables are load correctly. The same problem arise when loading for exemple x3d_ions__C1_CP_CIS_HIA_HS_MAG_IONS_PF.

The issue does not occur when I load the same cdf-file using an older version that I have lying around, v.1.7.1.

closed 2015-12-11 by vaivads

LoveAlm, 2015-12-14:
It seams that the fix has instead caused an error in importing the electron data. Only the first value of the energy table is loaded but repeated several times.

The df-plus and df-minus for the ion data is now a 31x1 matrix, with one element for each energy level. For the electron datasets df.plus and df.minus is Nx44 matrix, where N is the number of measurements and with 44 energy levels. The same is true for the energy tables dep_x{2,1}.data where the ions have a single energy table and the electrons a time series of energy tables. This is imported correctly by irf_get_data.

vaivads, 2015-12-14:
Please send example matlab lines that can reconstruct the problem.

LoveAlm, 2015-12-14:
I've tried this for several different data sets from 2007, 2008 and 2009, both C1 and C3. But as an example:

tint=[irf_time([2008 04 24 21 40 00]),irf_time([2008 04 24 22 10 00])];
edpf=irf_get_data(tint,'Data__C1_CP_PEA_PITCH_SPIN_DPFlux','caa','mat');

Which results in a edpf.dep_x{2,1}df.plus/minus which is a 171x1 matrix which the length of the time series repeating only the first value in the df.plus/minus table.I believe the raw data is supposed to be a Nx44 matrix (time x energy bins).

idpf=irf_get_data('Differential_Particle_Flux__C1_CP_CIS-HIA_PAD_HS_MAG_IONS_PF','caa','mat');

results in that idpf.dep_x{2,1}df.plus/minus which is a 31x1 matrix with one entry per energy level but no time series (energy bin x time), as is generally the case for CIS.

commit 7966f682 (2015-12-11): @dataobj.getmat: order in df.plus and df.minus changed after recent NASA CDF updates, fixes #23
commit b70d1834 (2015-12-15): @dataobj.getmat: closes #23