irfu/irfu-matlab #24: irf_resamp issues with MMS dfg data

elineriksson, 2015-12-04:
Using matlab2014b (with devel branch) I keep getting an error with the resamp function when trying to use dfg data.

Example run:

tint  = irf.tint('2015-11-28T00:00:00Z/2015-11-28T02:30:00Z');
c_eval('B?=mms.db_get_ts(''mms?_dfg_srvy_ql'',''mms?_dfg_srvy_dmpa'',tint);',[1:4]);
c_eval('B? = B?.resample(B1);',[2:4]);

Error I get:

Error using griddedInterpolant
The grid vectors are not strictly monotonic increasing.

Error in interp1 (line 183)
        F = griddedInterpolant(Xext,V,method);

Error in irf_resamp (line 234)
  out = [t interp1(x(:,1),x(:,2:end),t,method,'extrap')];

Error in TSeries/resample/resample_ (line 888)
        newData = irf_resamp([tData
        dataTmp],NewTime-TsTmp.time(1),varargin{:});

Error in TSeries/resample (line 875)
            case {'xy','xyz'}, resample_(obj); return

Error in TSeries/subsref (line 291)
          [varargout{1:nargout}] = builtin('subsref',obj,idx);

Error in c_eval (line 40)
if strfind(ev_str,'?'),

thomas-nilsson-irfu, 2015-12-04:
This issue is traced back to the source cdf files which have uneven timestamp and some which are barely increasing. Take for instance, "mms1_dfg_srvy_ql_20151128_v1.8.3.cdf", which when switching mode Fast to Slow, dumped raw have Epoch looking like this:

 Record # 1991: 2015-11-28T00:04:07.657261907
  Record # 1992: 2015-11-28T00:04:07.719762772
  Record # 1993: 2015-11-28T00:04:07.719769401
  Record # 1994: 2015-11-28T00:04:07.782263638
  Record # 1995: 2015-11-28T00:04:07.844764503
  Record # 1996: 2015-11-28T00:04:07.844771072
  Record # 1997: 2015-11-28T00:04:07.907265369
  Record # 1998: 2015-11-28T00:04:07.969766234
  Record # 1999: 2015-11-28T00:04:07.969772744
  Record # 2000: 2015-11-28T00:04:08.032267100
  Record # 2001: 2015-11-28T00:04:08.094767965
  Record # 2002: 2015-11-28T00:04:08.094774416
  Record # 2003: 2015-11-28T00:04:08.157268831
  Record # 2004: 2015-11-28T00:04:08.219769696
  Record # 2005: 2015-11-28T00:04:08.219776088
  Record # 2006: 2015-11-28T00:04:08.282270562
  Record # 2007: 2015-11-28T00:04:08.344771427
  Record # 2008: 2015-11-28T00:04:08.344777760
  Record # 2009: 2015-11-28T00:04:08.407272293
  Record # 2010: 2015-11-28T00:04:08.469773158

(This file had a minimum of 9 ns between two samples).

thomas-nilsson-irfu, 2015-12-07:
One fix was just added to MMSdevel, and when this commit has been merged to master this issue will automatically be closed by git.

Regarding the source files (DFG srvy ql), I received a reply stating:
"There is an interval of overlap where both Fast and Slow AFG and DFG data are produced. This overlap is included in L1B and QL data. It is not included in L2pre data."

commit d0a37f6c (2015-12-07): TSeries hopefully this fixes #24 subtract using int64 values, then convert to seconds (double precision).
closed 2016-02-12 by thomas-nilsson-irfu