irfu/irfu-matlab #28: mms.db_get_ts fails to load dis-dist data when needs to read from 3 data files

sergiotr, 2016-04-22:
Hi,
I am trying to load dis-dist data for a large interval (6 hours). dis-dist cdf files (fpi_fast_l2_dis-dist) cover 2 hours each. If my interval tint becomes larger than 4 hours, the script gives an error. Please note that dis-dist data is a 4-D matrix (time, energy, phi, theta).

to reproduce the error try to load 6 hours (21600 s) of data, interval 08:00 - 14:00 UT
tint = irf.tint('2015-09-21T08:00:00Z',21600);
disti=mms.db_get_ts('mms1_fpi_fast_l2_dis-dist','mms1_dis_dist_fast',tint);%this does not work (6 hours)

However, loading separately the intervals 08:00 - 12:00 UT or 10:00 - 14:00 UT does work, meaning the cdf files are not corrupted. E.g.:
tint = irf.tint('2015-09-21T08:00:00Z',14400);
disti1=mms.db_get_ts('mms1_fpi_fast_l2_dis-dist','mms1_dis_dist_fast',tint); %this works (4 hours)

tint = irf.tint('2015-09-21T10:00:00Z',14400);
disti2=mms.db_get_ts('mms1_fpi_fast_l2_dis-dist','mms1_dis_dist_fast',tint); %this works (4 hours)

I have been looking at +mms/mms_db.m but I cannot see the problem. Any clue?

Thanks 
Sergio

thomas-nilsson-irfu, 2016-04-22:
Dear Sergio,

could you please copy the error message you get.

I was unable to reproduce the error... (I removed some blank lines, but you clearly see the successful result below).

> tint = irf.tint('2015-09-21T08:00:00Z',21600)
> tint ...  EpochTT object : 2 records
>   2015-09-21T08:00:00.000000000Z
>   2015-09-21T14:00:00.000000000Z
> 
> disti=mms.db_get_ts('mms1_fpi_fast_l2_dis-dist','mms1_dis_dist_fast',tint)
> loading datastore...
>   [warning: variable2ts(57)] Times shifted to center of dt-+. dt-+ are recalculated
> 
> disti ... TSeries with properties:
>                 data: [4-D single]
>                 time: [4799x1 EpochTT]
>     coordinateSystem: ''
>          tensorOrder: 0
>          tensorBasis: ' ()'
>       representation: {[4800x1 EpochTT]}
>                 name: 'mms1_dis_dist_fast'
>                units: 'phase-space density'
>         siConversion: ''
>             userData: [1x1 struct]
> 
> size(disti.data) ... [ 4799          32          32          16]
> size(disti.time) ... [4799           1]

With best regards,
Thomas

sergiotr, 2016-04-22:
Waw, in my laptop at home it also works!
OK so it must be a local configuration problem of my database at the office
Thanks for checking and for your time.
Sergio

closed 2016-04-22 by sergiotr

neetasha, 2021-02-24:
Hello sir,
I am very new to space plasma field as well as data analysis work. Thus, I might be asking silly questions so sorry for the inconvenience.
I am using the function mms.db_get_ts ();
and this error is occuring "irf.log('fcal','nothing to substitute');"
Please will you let me know what this error is about. 

Thanks 
Neetasha

neetasha, 2021-02-25:
> Dear @neetasha <https://github.com/neetasha>,
> thanks for reaching out to us regarding your problem.
>
> However I am not entirely sure your problem is directly related to this
> old issue here, may I ask that you create a new issue here on
> https://github.com/irfu/irfu-matlab/issues/new/choose and fill in the
> information requested in the template (including but not limited to any
> commands you run in order to replicate your issue or more detailed* log
> outputs).
>
> *The error irf.log('fcal','nothing to substitute'); could potentially be
> called from different locations in the code, but currently it is not found
> in the mms specific sections of the irfu-matlab code (i.e. in subfolder
> mission/mms).

Hello
sir,
I have cleared the issue myself.
Thanks for the reply.
 Sir, I want to know more about the usage of the functions given in the
irf-matlab devel package.
So, are there any video lectures or websites through which I can learn
about various functions of the irf-matlab devel package and apply those for
data analysis of mms spacecraft.

Thanks and regards,
Neetasha Arya.

thomas-nilsson-irfu, 2021-02-25:
Dear @neetasha,
I'm pleased to hear your issue was resolved!

As for your second question, we unfortunately do not have any video tutorials or such.
What we do have for new users is mainly some example code, such as the various computing and plotting code found in the sub-folders `plots/` and `plots/mms/` and a semi-interactive `irf('demo');`. The results of some of these examples are briefly shown here: https://github.com/irfu/irfu-matlab/wiki#examples (it is mostly for Cluster but should be applicable with minor modifications to MMS as well). And of course we strive to have each m-functions header/help text describe the usage of each function.

Just as a minor point of order, may I ask that you, if you would not mind, open up a new issue if you encounter some problems in your usage of irfu-matlab or difficulty in locating appropriate function(-s) instead of replying here as it would help to reduce the amount of notifications sent to any user following this old (closed) issue. Thanks in advance.

With best regards,
Thomas Nilsson

neetasha, 2021-02-25:
Thank you for your response sir.
I am extremely sorry for the inconvenience and  I will make sure that I
create a new issue.

Regards,
Neetasha.