irfu/irfu-matlab #32: FPI v3.1.0 changes, mms.make_pdist quickfix

alexandros-lpp, 2016-10-26:
FPI v3.1.0 data breaks a few things in mms..make_pdist since energy0, energy1 and stepTable are no longer used.

Here is a suggested fix of mms..make_pdist.m
[make_pdist_fpi_v3_quickfix.zip](https://github.com/irfu/irfu-matlab/files/554621/make_pdist_fpi_v3_quickfix.zip)
It loads the new variable names for v3.1.0 and works as before for previous versions.
However, for v3.1.0 data the PDist methods that call the obsolete fields (energy0, energy1 and stepTable) will still have issues.

thomas-nilsson-irfu, 2016-10-27:
Dear @alexandros-lpp thank you for your work!

We are currently working on implementing it and the other fixes required (in various methods etc) for the new FPI v3 files.

alexandros-lpp, 2016-10-27:
Actually, since stepTable is still there in FPI v3 files, one can pull energy0 and energy1 from the first and second column of energy (assuming they don't change):

stepTable = stepTable.data;
energy0=energy(1+stepTable(1),:);
energy1=energy(1+stepTable(2),:);

Its not pretty or stable but sort of fixes some PDist methods.

closed 2016-12-19 by chasapis