irfu/irfu-matlab #40: It said that the IRBEM is not OK

Fenyutanchan, 2017-09-29:
irfu-matlab version:  v1.13.2
Checking if you have latest irfu-matlab... YES:)
IRBEM .. not OK. Please, contact irfu!
Operating system is OK
Matlab version is OK
CDF_LEAPSECONDSTABLE is OK

matlab version= '9.2.0.538062 (R2017a)'
System version is Microsoft Windows Version 10.0.15063

thomas-nilsson-irfu, 2017-09-29:
Dear @Fenyutanchan,

thank you for reporting this issue. 
The IRBEM package (included in [contrib/libirbem](https://github.com/irfu/irfu-matlab/blob/master/contrib/libirbem) originally from https://sourceforge.net/projects/irbem/) have presently only compiled support for unix system (that is Linux and Mac). If you do not presently need the IRBEM-LIB package, then this warning can be ignored and irfu-matlab can be used anyhow.

I will try to look into adding compiled mex files for Windows as well (from the Fortran code in IRBEM-LIB), or if that is unsuccessful then simply disable trying to run our checks of IRBEM on Windows machines.

With best regards,

Thomas Nilsson

Fenyutanchan, 2017-09-29:
@thomas-nilsson-irfu Thank you very much.

commit c94e5b95 (2017-09-29): Closes #40
Note: For most of irfu-matlab usage the IRBEM package is NOT required and the warning could be ignored!

I was not able to (easily) resolve the dependencies for the IRBEM, Matlab support no free fortran compiler for Windows and the provided "onera_desp_lib_Win64_x86.dll" library have a lot of dependencies (potentially a full install of MS Visual Studio and its redistributables could be required). Users are encuraged to configure it themselves, if required.
closed 2017-09-29 by Fenyutanchan