irfu/irfu-matlab #47: About mms data reading

yangz1984, 2019-04-18:
// TODO (you): please remove comments like this line and fill in this form as best you can, checking relevant boxes, etc.

### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).
Yes.

### Step 2: Describe your environment
 * irfu-matlab branch (i.e. "master", "devel"): v1.14.2
 * Matlab version used (i.e. "Matlab R2017a"): matlab R2018a
 * Operating system (ie "Windows 7", "Mac OS X 10.12.3", "Linux Ubuntu 16.04"): Windows 10
 - [ ] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem
**When I use "get_data" subroutine for the mms mission.**
>> Tint = irf.tint('2015-09-21T00:00:00Z/2015-09-21T17:00:00Z');
>> V1 = mms.get_data('V_gse',Tint,1);
**It reports:**
  [warning: mms_db.list_files(35)] No databases initialized
  [warning: mms_db.list_files(35)] No databases initialized
  [warning: mms_db.list_files(35)] No databases initialized
  [warning: mms_db.list_files(35)] No databases initialized
  [warning: mms_db.list_files(35)] No databases initialized

#### Expected behavior
Download the data automatically and get the V_gse data from a defined time range.

#### Actual behavior
Report "No databases initialized"

#### Steps to reproduce:
  // TODO (you): describe how to reproduce the problem. (Or if you prefer: you can fill in the "Relevant code" section below).
  1. Tint = irf.tint('2015-09-21T00:00:00Z/2015-09-21T17:00:00Z');
  2. V1 = mms.get_data('V_gse',Tint,1);

#### Relevant code:

  ```
    // TODO (you): insert code here to reproduce the problem.
    // (Or if you prefer: you can descbibe it in the "Steps to reproduce" section above).
  ```

thomas-nilsson-irfu, 2019-04-18:
Dear @yangz1984, thank you for this issue report.

Presently the `mms.get_data()` function is used to locate and read data which is found on the local filesystem only (or mounted via `sshfs`, or similar functionality). In order to help define the path where it should look for data locally one must first initialize a database, using for instance:
`mms.db_init('local_file_db','/data/mms')` 
That is if the data folder structure is found under `/data/mms/` on the machine it is running on. 
The folder structure under this root should contain subfolders for each mms spacecraft (mms1, mms2, mms3, mms4), in the next level it should contain folder for each instrument (aspoc, edp, fpi, etc) and the next level down mode and so on, similar to the paths used at the Science Data Center at LASP.

@yurikhot and I have discussed about adding support for automatically downloading data from SDC to the users machine, unfortunately we have yet to implement such functionality.
If (or when) this gets properly implemented we must also have a way of ensuring that the user (like yourself) is not accentually made to download several terabytes of data from SDC (https://lasp.colorado.edu/mms/sdc/public/about/how-to/) or NASA's mirror (https://cdaweb.gsfc.nasa.gov/index.html/).

I do apologize for the inconvenience.

yangz1984, 2019-04-19:
Thanks for the information.

yangz1984, 2019-04-19:
Dear Thomas,

(1) I download the data "mms1_fpi_brst_l2_dis-dist_20160111053744_v3.3.0.cdf" and put it in the following path: C:\Users\zwyang\Desktop\data\mms\mms1\fpi\brst\l2\dis-dist\2016\01\11

(2) In matlab, type irf. It seems ok. Then I use 
>>mms.db_init('local_file_db','C:\Users\zwyang\Desktop\data\mms')" 
to add the path. Done.

(3) Type command as below:
>> Tint = irf.tint('2016-01-11T05:37:44Z/2016-01-11T05:38:00Z');
>> ni = mms.get_data('Ni_fpi_brst_l2',Tint,1);
It reports: [warning: get_data/get_ts(611)] No data for mms1_fpi_brst_l2_dis-moms(mms1_dis_numberdensity_brst)

Could you help check what is wrong, please? Thank you very much!

chasapis, 2019-04-19:
the file you downloaded is the particle distributions ("dis-dist"). You need to download the moments ("dis-moms") to read the density. So the filename you'd look for would be "mms1_fpi_brst_l2_dis-moms_20160111053744_v3.3.0.cdf"

yangz1984, 2019-04-19:
It works. Thanks a lot.

yangz1984, 2019-04-22:
Dear Chasapis,
One more question please. When we load the data in the SPEDAS, it list all structures we can read (please see below). E.g., 
"c:\data\mms\mms1\fpi\brst\l2\dis-moms\2016\01\11\mms1_fpi_brst_l2_dis-moms_20160111053744_v3.3.0.cdf"
STORE_DATA(261): Creating tplot variable: 55 mms1_dis_errorflags_brst
STORE_DATA(261): Creating tplot variable: 56 mms1_dis_compressionloss_brst
STORE_DATA(261): Altering tplot variable: 1 mms1_dis_steptable_parity_brst
STORE_DATA(261): Altering tplot variable: 2 mms1_dis_startdelphi_count_brst
STORE_DATA(261): Altering tplot variable: 3 mms1_dis_startdelphi_angle_brst
STORE_DATA(261): Altering tplot variable: 8 mms1_dis_sector_despinp_brst
STORE_DATA(261): Altering tplot variable: 19 mms1_dis_energyspectr_px_brst
STORE_DATA(261): Altering tplot variable: 20 mms1_dis_energyspectr_mx_brst
STORE_DATA(261): Altering tplot variable: 21 mms1_dis_energyspectr_py_brst
STORE_DATA(261): Altering tplot variable: 22 mms1_dis_energyspectr_my_brst
STORE_DATA(261): Altering tplot variable: 23 mms1_dis_energyspectr_pz_brst
STORE_DATA(261): Altering tplot variable: 24 mms1_dis_energyspectr_mz_brst
STORE_DATA(261): Altering tplot variable: 25 mms1_dis_energyspectr_omni_brst
STORE_DATA(261): Altering tplot variable: 26 mms1_dis_spectr_bg_brst
STORE_DATA(261): Altering tplot variable: 27 mms1_dis_numberdensity_bg_brst
STORE_DATA(261): Altering tplot variable: 28 mms1_dis_numberdensity_brst
STORE_DATA(261): Altering tplot variable: 29 mms1_dis_numberdensity_err_brst
etc.

E.g., How to choose "mms1_dis_energyspectr_omni_brst" for plotting in irfu-matlab? I did not find it in keywords of the subroutine "get_data.m". Thank you very much!

yangz1984, 2019-04-22:
My clumsy way is to use the following command read Px, Mx, Py, My, Pz, Mz, respectively and make an average to get the omni energyspectr. Please let me know if irfu-matlab can do it better. Thanks a lot.
res=mms.db_get_ts('mms1_fpi_brst_l2_dis-moms','mms1_dis_energyspectr_px_brst',Tint);

yurikhot, 2019-04-23:
пн, 22 квіт. 2019 о 11:20 yangz1984 <notifications@github.com> пише:

> Dear Chasapis,
> One more question please. When we load the data in the SPEDAS, it list all
> structures we can read (please see below). E.g., [...]

See Example_MMS_particle_DEFlux.m and Example_MMS_particle_PAD.m

Yuri

-- 
Associate Prof. Yuri Khotyaintsev
Institutet för rymdfysik (IRF), Uppsala
tel: +46 18 471 59 29
fax: +46 18 471 59 05

thomas-nilsson-irfu, 2019-06-14:
Closing this issue, if problem persists: feel free to re-open or submit a new issue.

closed 2019-06-14 by thomas-nilsson-irfu