irfu/irfu-matlab #49: Components of TSeries should have tensorOrder 0

ajohlander, 2019-06-14:
### Step 1: Latest code?
- [x] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

### Step 2: Describe your environment
 * irfu-matlab branch (i.e. "master", "devel"): devel
 * Matlab version used (i.e. "Matlab R2017a"): R2018a
 * Operating system (ie "Windows 7", "Mac OS X 10.12.3", "Linux Ubuntu 16.04"): OSX 10.14.5
 - [x] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem

#### Expected behavior
A component of a time series object should return a time series object with the variable "tensorOrder" set to 0, i.e. a scalar .

#### Actual behavior
Components of TSeries objects inherit the tensor order of the original objects. True for both vectors (i.e. E.x) and tensors (i.e. Ti.xx).

This is annoying when trying to add time series together.

#### Steps to reproduce:
#### Relevant code:
>> tint = irf.tint('2017-10-21T21:29:30/2017-10-21T21:31:30');
>>E = mms.get_data('E_gse_edp_fast_l2',tint,1);
>>Ex = E.x;
>> Ex.tensorOrder

ans =

     1

#### WHAT I TRIED

I tried just forcing tensorOrder to 0 in TSeries.getComponent (lines: 1529 & 1559) but this causes an error: 

Error using TSeries (line 247)
Representation already set for all DATA dimensions

I don't know what to do about that.

thomas-nilsson-irfu, 2019-06-14:
Dear @ajohlander,
thanks for letting us know of this issue.

I am not sure exactly how you tried to force tensorOrder to 0 but the TSeries function would require some more changes than simply changing line 1529 to have `'TensorOrder',0` (or `teno-1`) this as the TSeries created on line 1535 also have arguments regarding `representation` and `TensorBasis` of this E.x scalar. A scalar TSeries() is created if no TensorBasis or representation is given, with `'TensorOrder', 0`, see for instance irf.ts_scalar().

https://github.com/irfu/irfu-matlab/blob/f34a102eaf15a39a4923dd14bd923fdaf4f007ec/TSeries.m#L1529-L1535

This code segment should perhaps be changed to not add TensorBasis and loop through adding `repres` for the special case of `nd==1`...

thomas-nilsson-irfu, 2019-06-14:
Having looked at the code for TSeries.m, I found something looking rather strange.. 
(I am writing it here mainly to keep track, hope you do not mind!).

Just above the code which is causing you problem, `nd` is computed:
https://github.com/irfu/irfu-matlab/blob/f34a102eaf15a39a4923dd14bd923fdaf4f007ec/TSeries.m#L1499
and then we have:
https://github.com/irfu/irfu-matlab/blob/f34a102eaf15a39a4923dd14bd923fdaf4f007ec/TSeries.m#L1514-L1528
Especially note the `case 5` where we must have had `ndims(obj.data_) == 6`...

This problem with dimensions appears to have appeared elsewhere as well (but here it should not matter that much, as higher dimensions, value 1, are always included regardless):
https://github.com/irfu/irfu-matlab/blob/f34a102eaf15a39a4923dd14bd923fdaf4f007ec/TSeries.m#L1461-L1468