irfu/irfu-matlab #51: About the plot of plasma pitch angle distribution_MMS

zhang-chi-IGGCAS, 2020-06-08:
### Step 1: Latest code?

ic = 1; % Spacecraft number
Tint = irf.tint('2017-06-17T20:20:00.00Z/2017-06-17T20:30:00.00Z');
c_eval('ePDist_brst = mms.get_data(''PDe_fpi_fast_l2'',Tint,?);',ic);
c_eval('Bxyz=mms.db_get_ts(''mms?_fgm_srvy_l2'',''mms?_fgm_b_gsm_srvy_l2'',Tint);',ic);
ePDistpitch = ePDist.e64.pitchangles(Bxyz,24).deflux;

### Step 2: Describe your environment

Windows   Matlab 2019b

### Step 3: Describe the problem

wrong use:  TSeries (line 119)
T and DATA must have the same number of records

wrong: PDist/e64 (line 2138)
      [pdistr,phir,energyr] =
      mms.psd_rebin(obj,TSeries(obj.time,obj.depend{2}),obj.ancillary.energy0,obj.ancillary.energy1,TSeries(obj.time,obj.ancillary.esteptable));

wrong: PDist/subsref (line 99)
        [varargout{1:nargout}] = builtin('subsref',obj,idx);

#### Actual behavior

when i use the electron data in burst mode, the pitch angle distrubution of electron can be well obtained. However, when i use the fast mode data, it would be a error. I tried to find the mistake,  I found that the size of ePDist.depend_2  is  [1*32] in fast mode, while is  [length(time), 32] in burst mode. so could you help me solve this problem?  Thanks a lot

danbgraham, 2020-06-08:
I've added a minor fix to the function which calculates the pitch angle distribution so it should work with fast mode data. This is added to the devel branch.

Some points about the command lines used: 
(1) The function e64 recalculates the particle distribution 64 energy channel resolution and half the time resolution. It will only work when there are two energy tables used. This means it will only work for burst mode data in the early phase of the mission. It will not work for fast mode data and in general is not needed for most analyses. 
(2) When calculating the pitch angle the B field must be in the same coordinate system as the particle distribution function (dmpa coordinates for B). 

E.g.,
ic = 1;
Tint = irf.tint('2017-06-17T20:20:00.00Z/2017-06-17T20:30:00.00Z');
c_eval('ePDist = mms.get_data(''PDe_fpi_fast_l2'',Tint,?);',ic);
c_eval('Bxyz=mms.get_data(''B_dmpa_fgm_srvy_l2'',Tint,?);',ic);
ePDistpitch = ePDist.pitchangles(Bxyz,24).deflux;
irf_spectrogram(ePDistpitch.specrec('pitchangle'))

zhang-chi-IGGCAS, 2020-06-09:
Now it can work well, it is very helpful for me , thank a lot!!

zhang-chi-IGGCAS, 2020-06-09:
Thanks a lot, it is very helpful for me.  this command line  
"ePDistpitch = ePDist.pitchangles(Bxyz,24).deflux;"  can work very well, and we can get the variable which is named as "ePDistpitch"; 
However, when i want to plot,  such a mistake has occurred. I used this command line:

eint1 = [3e1 3e2];
irf_spectrogram(h,ePDistpitch.elim(eint1).specrec('pa'),'log','donotfitcolorbarlabel');

Then matlab shows that"
wrong use ismember>ismemberR2012a (line 212)
wrong ismember (line 148)
            [lia,locb] = ismemberR2012a(A,B,logical(flaginds(1)));
wrong PDist/elim (line 2229)
          [esteptable,~] =
          ismember(energy,PD.ancillary.energy1,'rows');
wrong PDist/subsref (line 123)
          [varargout{1:nargout}] = builtin('subsref',obj,idx);

I think that it is due to the "ePDist.ancillary" is empty, so could you help me to solve it?

zhang-chi-IGGCAS, 2020-06-09:
Thanks a lot, it is very helpful for me. this command line
"ePDistpitch = ePDist.pitchangles(Bxyz,24).deflux;" can work very well, and we can get the variable which is named as "ePDistpitch";
However, when i want to plot, such a mistake has occurred. I used this command line:

eint1 = [3e1 3e2];
irf_spectrogram(h,ePDistpitch.elim(eint1).specrec('pa'),'log','donotfitcolorbarlabel');

Then matlab shows that"
wrong use ismember>ismemberR2012a (line 212)
wrong ismember (line 148)
[lia,locb] = ismemberR2012a(A,B,logical(flaginds(1)));
wrong PDist/elim (line 2229)
[esteptable,~] =
ismember(energy,PD.ancillary.energy1,'rows');
wrong PDist/subsref (line 123)
[varargout{1:nargout}] = builtin('subsref',obj,idx);

I think that it is due to the "ePDist.ancillary" is empty, so could you help me to solve it? 

Actually, I just want to plot the ion pitch angel distribution during about 1 hour interval, so I have to use fast-mode data. Could you show me a example about how to do it ?

danbgraham, 2020-06-09:
We've made a revision that should correct this now. Please try again with the latest version. 

Regarding, an example of plotting fast mode data, you can use the latest version of Example_MMS_particle_PAD.m. For the initial loading of variables you can replace brst with fast (for FPI and EDP products) and brst with srvy (for FGM products). The analyses and plotting should now work fine with both fast and brst data.

zhang-chi-IGGCAS, 2020-06-09:
Thanks a lot ,it can work well now.

closed 2020-06-09 by zhang-chi-IGGCAS