irfu/irfu-matlab #58: How to plot the electron velocity distribution?

zhang-chi-IGGCAS, 2020-09-12:
Dear all:
        I am very sorry to bother you, I am a student, and don't know how to plot the electron velocity distribution, my code is :

tint=irf.tint('2017-07-11T22:34:02Z/2017-07-11T22:34:04Z');
ic=1;
c_eval('B=mms.get_data(''B_dmpa_fgm_srvy_l2'',tint,?);',ic);  %magnetic field 

%electric field and electron data
c_eval('E=mms.get_data(''E_dsl_edp_brst_l2'',tint,?);ne=mms.get_data(''Ne_fpi_brst_l2'',tint,?);',ic);  
c_eval('ePDist=mms.get_data(''PDe_fpi_brst_l2'',tint,?);',ic);

%spacecraft Pot
c_eval('SCpot=mms.db_get_ts(''mms?_edp_brst_l2_scpot'',''mms?_edp_scpot_brst_l2'',tint);',ic);

tint=irf.tint('2017-07-11T22:34:02.367Z/2017-07-11T22:34:02.397Z'); dist=ePDist.tlim(tint); time=irf_time(mean(irf_time(tint,'epochtt>datenum')),'datenum>epochTT');  %the mean time
n0=ne.resample(time).data;  E0=E.resample(time).data;   B0=B.resample(time).data;    sc=SCpot.resample(time).data;
hatB0 = double(irf_norm(B0));   hatE0 = double(irf_norm(E0));   hatExB0 = cross(hatE0,hatB0);
 hatE0 = -cross(hatExB0, hatB0);
xyz =[hatB0; hatExB0; hatE0; ];
mms.plot_projection(h1,ePDist,'tint',tint,'xyz',xyz, 'scpot',scpot,'vlabel',vlabels);
colormap(h1,'jet');

but  the result as shown in this figure is different to Figure 2K in Torbert et al., 2018, science.  doi: 10.1126/science.aat2998
I don't know why, so could you help me? thank you

danbgraham, 2020-09-12:
Hi, 

There are a couple of reasons why the distribution is different. mms.plot_projection plots a slice of the 3D distribution, whereas in the Torbert paper they are plotting a 2D reduced distribution, meaning the third velocity axis is integrated over (note that the units are different between the two methods). It can look different because you are plotting something different to the Torbert paper. Also in your script you are using MMS1 data, whereas Torbert et al. is using MMS3 data, so this may also account for a different looking distribution. 

Here is an example of the 2D reduced distribution seen in Torbert et al. (see also Example_MMS_reduced_ele_dist_2D.m): 
vg = linspace(-40e3,40e3,100); % km/s
nMC = 2e3;
f2D = dist.reduce('2D',hatB0,hatExB0,'base','cart','vg',vg,'nMC',nMC,'lowerelim',100);

fn=figure;
set(fn,'Position',[10 10 400 350])
h(1)=axes('position',[0.12 0.13 0.8 0.8]); % [x y dx dy]
ud=get(fn,'userdata');
ud.subplot_handles=h;
set(fn,'userdata',ud);
set(fn,'defaultLineLineWidth',1); 

    f2D.plot_plane(h(1),'docolorbar',0);
    caxis(h(1),[-14 -10]);
    hcb = colorbar(h(1));
    ylabel(hcb,'log_{10} f_e (s^2 m^{-5})','interpreter','tex','fontsize',12)
    hold(h(1),'on')
    plot(h(1),[0 0 ],max(abs(vg))*[-1 1],'color','k','linewidth',1)
    plot(h(1),max(abs(vg))*[-1 1],[0 0 ],'color','k','linewidth',1)
    hold(h(1),'off')
    grid(h(1),'off')
    axis(h(1),'equal')
    colormap(h(1),'jet')
    irf_legend(h(1),'(c)',[0.98 0.85],'color','w','fontsize',12)
    axis(h(1),1e-3*[min(vg) max(vg) min(vg) max(vg)])
    xlabel(h(1),'V_{||} (10^3 km s^{-1})','interpreter','tex','fontsize',12)
    ylabel(h(1),'V_{\perp} (10^3 km s^{-1})','interpreter','tex','fontsize',12)
    set(gcf,'color','w')

zhang-chi-IGGCAS, 2020-09-14:
thanks a lot!

closed 2020-09-14 by zhang-chi-IGGCAS