irfu/irfu-matlab #66: Library does not load MacOS

gmisha59, 2021-01-09:
### Step 1: Latest code?
- [ x] This issue is present in the most recent code v 1.15.1 (downloaded as zip)

### Step 2: Describe your environment
 * irfu-matlab branch (i.e. "master", "devel"): master 1.15.1
 * Matlab version used (i.e. "Matlab R2017a"): R2020b
 * Operating system (ie "Windows 7", "Mac OS X 10.12.3", "Linux Ubuntu 16.04"): MacOS 11.1
 - [x ] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem

#### Expected behavior
irf run complete

#### Actual behavior
#### Relevant code:
>> irf
Added to path: /Users/gedalin/irfu-matlab-master/
Added to path: /Users/gedalin/irfu-matlab-master/contrib/isdat
Added to path: /Users/gedalin/irfu-matlab-master/contrib/libirbem
Added to path: /Users/gedalin/irfu-matlab-master/contrib/libcef
Added to path: /Users/gedalin/irfu-matlab-master/contrib/matlab_central
Added to path: /Users/gedalin/irfu-matlab-master/contrib/matlab_central/cm_and_cb_utilities
Added to path: /Users/gedalin/irfu-matlab-master/contrib/mice
Added to path: /Users/gedalin/irfu-matlab-master/contrib/nasa_cdf_patch
Added to path: /Users/gedalin/irfu-matlab-master/irf
Added to path: /Users/gedalin/irfu-matlab-master/plots
Added to path: /Users/gedalin/irfu-matlab-master/plots/mms
Added to path: /Users/gedalin/irfu-matlab-master/mission/cluster
Added to path: /Users/gedalin/irfu-matlab-master/mission/cluster/caa
Added to path: /Users/gedalin/irfu-matlab-master/mission/solar_orbiter
Added to path: /Users/gedalin/irfu-matlab-master/mission/solar_orbiter/bicas/src
Added to path: /Users/gedalin/irfu-matlab-master/mission/themis
Added to path: /Users/gedalin/irfu-matlab-master/mission/thor
Added to path: /Users/gedalin/irfu-matlab-master/mission/mms
Added to path: /Users/gedalin/irfu-matlab-master/mission/mms/mms_testFunctions
irfu-matlab version:  v1.15.1
Checking if you have latest irfu-matlab... YES:)
Error using loadlibrary
There was an error loading the library
"/Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib"
dlopen(/Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib, 6): no suitable image
found.  Did find:
	/Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib: code signature in
        (/Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib) not valid for use
        in process using Library Validation: library load disallowed by system policy

Error in cef_init (line 6)
	loadlibrary (['libcef_' lower(computer)], @libcef_mfile,'alias','libcef')

Error in irf (line 234)
        cef_init();

Error in irf (line 42)
  irf('ceflib');

Caused by:
    Error using loaddefinedlibrary
    dlopen(/Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib, 6): no suitable
    image found.  Did find:
    	/Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib: code signature in
        (/Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib) not valid for use
        in process using Library Validation: library load disallowed by system policy

thomas-nilsson-irfu, 2021-01-09:
Dear @gmisha59,

thanks for taking the time to report this issue. I will just write a short message here for now, hopefully I can help fix it better on Monday during working hours (CET).

Looking at the reported problem (`code signature in (libcef_maci64.dylib) not valid for use in process using Library Validation: library load disallowed by system policy`) it appears to be related to Apple's Mac OS protection. This is preventing execution of compiled binary code and compiled libraries which comes from the internet.

It appears to be fixable in a few ways:
1) Disable the protection (quarantine) for this library file. One google search lead me to https://forum.blackmagicdesign.com/viewtopic.php?f=12&t=106755 which describes the command required as, in your case:
`sudo xattr -d com.apple.quarantine /Users/gedalin/irfu-matlab-master/contrib/libcef/libcef_maci64.dylib`

or

2) Disable the loading of this library (if your intended usage of irfu-matlab does not require the libcef library, used for Cluster Exchange Format/CEF files), simply comment out the line in question in your local downloaded copy of [irf.m#L42](https://github.com/irfu/irfu-matlab/blob/7d12bf7162a6b2c19316a6599f7c486989dc40b7/irf.m#L42). This should prevent trying to load the library in question , however may result in a similar error appearing for loading other libraries (mice, irbem, etc.) which were not reached the first time you ran `irf;`.

or

3) Compile the code yourself in the https://github.com/irfu/irfu-matlab/tree/master/contrib/libcef directory. (Download from http://ceflib.irap.omp.eu/page=download).

- Side note:
In the worst case, it is a Matlab compatibility issue as reported in https://se.mathworks.com/matlabcentral/answers/467753-why-do-i-get-a-bundle-60-start-failed-error-using-imsegkmeans-on-macos-10-10-yosemite
Not exactly the same message, but officially Matlab R2020b does not explicitly state support for MacOS 11.1 (only 11 flat), as per https://se.mathworks.com/support/requirements/matlab-system-requirements.html.

With best regards,

Thomas Nilsson

thomas-nilsson-irfu, 2021-01-11:
Dear @gmisha59,

I have made one change to the latest development version of our code (available https://github.com/irfu/irfu-matlab/tree/devel), which should no longer result in a crash for a failure to load the `libcef_maci64.dylib` file. This change will be included in our next release of irfu-matlab.

I will close this issue here and now. You can download the development version as a zip here: https://github.com/irfu/irfu-matlab/archive/devel.zip, or preferably I would suggest you use the [git clone](https://docs.github.com/en/free-pro-team@latest/github/creating-cloning-and-archiving-repositories/cloning-a-repository) feature so you could easily update it with the latest code using a simple `git pull` command.

If you still experience problems feel free to get back to us.

With best regards,

Thomas Nilsson

closed 2021-01-11 by thomas-nilsson-irfu