irfu/irfu-matlab #71: MMS Energyspectrum

01344240, 2021-04-06:
Hi Everyone,
Is there a way to make plots of the ion/electron energyspectrum plots for MMS?
Thanks,
Paul.

wyli1124, 2021-04-07:
Dear Paul,

         Firstly, you can load the ion distribution functions via:

       iDist = mms.get_data('PDi_fpi_brst_l2', Tint, 1);
          Here, Tint is the time interval, and 1 (not L) is the MMS
spacecraft you are interested.

         Then, you can make energy spectrum plots via:
          irf_spectrogram(iDist.deflux.omni.specrec, 'log');

Best regards,
Wenya

01344240, 2021-04-07:
> This email from ***@***.*** originates from outside Imperial. Do not click on links and attachments unless you recognise the sender. If you trust the sender, add them to your safe senders list<https://spam.ic.ac.uk/SpamConsole/Senders.aspx> to disable email stamping for this address.

Dear Wenya,

Thank you so much for this!

Paul.

thomas-nilsson-irfu, 2021-04-09:
I'm considering this issue fixed and closing it now.
If anyone disagrees, feel free to reopen this issue.

closed 2021-04-09 by thomas-nilsson-irfu