irfu/irfu-matlab #73: How to use whamp codes for plotting dispersion relations?

neetasha, 2021-04-26:
### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

### Step 2: Describe your environment
 * irfu-matlab branch (i.e. "master", "devel"): __devel__
 * Matlab version used (i.e. "Matlab R2017a"): __Matlab R2020a__
 * Operating system (ie "Windows 7", "Mac OS X 10.12.3", "Linux Ubuntu 16.04"): __Microsoft Windows 10 Pro_
 - [ ] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem
I want to use Whamp codes to plot wave dispersion. The irf-matlab devel package contains /irf-matlab/+whamp/, whamp codes. I want to plot dispersion relation for Lower hybrid waves. Will you please guide me how to plot this using whamp code? How to initiate it?

thomas-nilsson-irfu, 2021-04-26:
Dear @neetasha,
let me begin by stating that I am not too familiar with the `WHAMP` package itself... (I am currently in the process of reading the original report trying to understand the programs input and output parameters following the issue irfu/whamp#2 but have not yet found the time to properly dig into it).

That being said however, one good starting point could perhaps be the `WHAMP` quick reference document https://github.com/irfu/whamp/blob/master/doc/whamp_quick_reference/whamp_quick_reference.pdf from our https://github.com/irfu/whamp repository.

neetasha, 2021-04-26:
Sure, I will refer WHAMP quick reference document.
Thanks sir for the information.

thomas-nilsson-irfu, 2021-06-02:
I'm closing this issue, should you still have problems feel free to comment again or open a new issue.

closed 2021-06-02 by thomas-nilsson-irfu