irfu/irfu-matlab #74: How to start to analyse PSP data with irfu packages?

Krushna-100, 2021-04-29:
// TODO (you): please remove comments like this line and fill in this form as best you can, checking relevant boxes, etc.

### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

 Step 2: Describe your environment
 * irfu-matlab branch (i.e. "master", "devel"): _PSPdevel
 * Matlab version used (i.e. "Matlab R2017a"): _MATLAB R2018b
 * Operating system (ie "Windows 7", "Mac OS X 10.12.3", "Linux Ubuntu 16.04"): _Windows 10 (64 bit)
 - [ ] A 64-bit installation of Matlab was used. 

### Step 3: Describe the problem
I want to use the irfu-matlab package to analysis the PSP data. I am a novice in this field. I have no idea about how to start with the irfu-packages. I need help from the experts of this field. Your suggestions and help will be highly appreciated.

With best regards
Krushna 
#### Expected behavior

#### Actual behavior

#### Steps to reproduce:
  // TODO (you): describe how to reproduce the problem. (Or if you prefer: you can fill in the "Relevant code" section below).
  1. _____
  2. _____
  3. _____

#### Relevant code:

  ```
    // TODO (you): insert code here to reproduce the problem.
    // (Or if you prefer: you can descbibe it in the "Steps to reproduce" section above).
  ```

thomas-nilsson-irfu, 2021-05-03:
Dear @Krushna-100,

I will try to answer a bit of your question as for how to get started with `irfu-matlab`. But please note that I have not written (or looked in detail at) the code in our specific `PSPdevel` branch as it is very much a work in progress and it is under active development. I suspect some of the contributors of this PSP specific code may be able to help if you have some more specific questions.

As for getting started you should:
1. start Matlab (I assume you have a 64-bit install even though you did not tick this specific checkbox in our issue template).
2. change directory to your local copy of `irfu-matlab` (its root folder).
3. execute `irf;` to setup the required paths and environment.
4. then to load PSP data specifically, I would point you to the function [psp_load](https://github.com/irfu/irfu-matlab/blob/PSPdevel/mission/psp/psp_load.m) and its documentation/comments.

This `psp_load` function does appear to have some support for automatically downloading the PSP cdf files to a local copy on your machine if they are not present (may require some login) and then read them into our `TSeries` object.

vaivads, 2021-05-04:
Some examples how to read and plot the PSP data are here:
https://gist.github.com/vaivads/b2dff623fef85e2a8d0a35331ce156c9

It's very much work in progress, so please notify of any problems.

Thanks

thomas-nilsson-irfu, 2021-06-02:
I'm closing this issue, should you still have problems feel free to comment again or open a new issue.

closed 2021-06-02 by thomas-nilsson-irfu