irfu/irfu-matlab #81: Regarding LMN coordinates

neetasha, 2021-08-31:
### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

### Step 2: Describe your environment
 * irfu-matlab branch (e.g. ```master```, ```devel```): __'''devel'''__
 * Matlab version used (e.g. ```Matlab R2020b```): **__```Matlab R2020a```__**
 * Operating system (e.g. ```Windows 10```, ```Mac OS X 10.15.6```, ```Linux Ubuntu 20.04```): _```Windows 10```___
 - [ ] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem
I am using mms FPI data. I want to convert GSE/GSM coordinates into LMN coordinate using irfu-matlab.
Will you please guide me through this issue of converting XYZ into LMN.
Thanks in advance.
Neetasha Arya.

danbgraham, 2021-08-31:
Hi Neetasha, 

Which FPI data are you using? Are you referring to distribution functions or vector particle moments?

Regards,
Daniel

neetasha, 2021-08-31:
I am using particle moments from FPI . Also For Magnetic field and Electric field using FGM and EDP .
please see the attached files.
[B(nT).csv](https://github.com/irfu/irfu-matlab/files/7086246/B.nT.csv)
[V_i.csv](https://github.com/irfu/irfu-matlab/files/7086247/V_i.csv)

danbgraham, 2021-08-31:
You can use the function irf_newxyz.m to rotate vector data into a new coordinate system. You just need to specify the vectors of the LMN coordinate system.

neetasha, 2021-09-01:
Its working now. Thanks a lot.