irfu/irfu-matlab #86: Issue with running the "Example_MMS_Polarizationanalysis.m" code in Matlab

Krushna-100, 2022-01-28:
// TODO (you): please remove comments like this line and fill in this form as best you can, checking relevant boxes, etc.

### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

### Step 2: Describe your environment
 * irfu-matlab branch (e.g. ```master```, ```devel```): devel
 * Matlab version used (e.g. ```Matlab R2020b```): Matlab R2018b
 * Operating system (e.g. ```Windows 10```, ```Mac OS X 10.15.6```, ```Linux Ubuntu 20.04```): Windows 10
 - [ ] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem

I am trying to plot the wave characteristics of plasma waves in Matlab using irfu-matlab packages. I found the code "Example_MMS_Polarizationanalysis.m" useful, which is used to perform polarization analysis for for the burst mode Electric and Magnetic field data. I am eager to see what kind of results this code is giving. However, while running the code, I am getting an error message like "Reference to non-existent field _'gseR1'". I am unable to find this data from the CDA web. 

Can anybody from the expert committee give any suggestions how to resolve the issue?
Your early response will be highly appreciated.

Thanking you

With best regards
Krushna Chandra Barik
   

#### Expected behavior

#### Actual behavior

#### Steps to reproduce:
  // TODO (you): describe how to reproduce the problem. (Or if you prefer: you can fill in the "Relevant code" section below).
  1. _____
  2. _____
  3. _____

#### Relevant code:

  ```
    // TODO (you): insert code here to reproduce the problem.
    // (Or if you prefer: you can describe it in the "Steps to reproduce" section above).
  ```

thomas-nilsson-irfu, 2022-01-28:
Dear @Krushna-100,

as for your issue. I would prefer if you could copy the entire error message you get since it could help track down where the issue comes from.

Anyhow, I suspect it is the loading of position which fails in your instance with the code
https://github.com/irfu/irfu-matlab/blob/d0831c354372f4acc8a87ea7bf3a5ea657af4ebd/plots/mms/Example_MMS_Polarizationanalysis.m#L14-L15
(where `R_gse` is an alias we use in irfu-matlab for positional information in the GSE frame of reference).

The problem could be that you do not have the corresponding files located on your system, or perhaps not initialized the database correctly. Something similar occurred in issue #69, so please have a look there to see the discussion there can help.

With best regards,

Thomas Nilsson

Krushna-100, 2022-01-28:
Dear Thomas,

Thank you very much for your early response to my email. I will definitely go through issue #69 and let you know whether my problem has been resolved or not.

With best regards
Krushna Chandra Barik

Krushna-100, 2022-01-28:
Dear Thomas,

Thank you very much for your timely help. The issue that I have reported in my earlier email was sorted out. Now, I can produce very beautiful figures for the wave polarization properties using the irfu-matlab package, which is also my first ever plot using the same package.

Thank you so much once again for your kind support.

With best regards
Krushna Chandra Barik

closed 2022-01-31 by thomas-nilsson-irfu