irfu/irfu-matlab #87: mms.get_data returns wrong HPCA moments

ajohlander, 2022-02-28:
### Step 1: Latest code?
- [x] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

### Step 2: Describe your environment
branch: devel
Matlab: 2021b
Mac OS: 12.1
 - [x] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem
The function "mms.get_data" with variable string 'Nheplusplus_hpca_srvy_l2' should return the number density of alpha particles (He++) but instead returns the number density of He+.

Same issue seems to be present with all He2+ HPCA moments. Burst or survey mode does not make a difference. Distribution functions seem to work as expected.

Using e.g., 'Nheplus_hpca_srvy_l2' returns the correct He+ density. 

I checked the CDF file and the data there looks good. I would fix this myself but I am not familiar with how mms.get_data works.

#### Expected behavior
He2+ moments

#### Actual behavior
He+ moments

#### Steps to reproduce:
Tint = irf.tint('2018-01-23T08:00:00/2018-01-23T12:00:00');
Nhe2p = mms.get_data('Nheplusplus_hpca_srvy_l2',Tint,1)

cecilianorgren, 2022-02-28:
The problem is here:

    all_ions = {'hplus','heplus','heplusplus','oplus'};
    ion_index = cellfun(@(s) ~isempty(strfind(Vr.param, s)), all_ions);
ion = all_ions{find(ion_index)};

ion_index = [0 1 1 0]: i.e.it returns 1 for both heplus and heplusplus and ion becomes the first instance. I can fix it.

ajohlander, 2022-02-28:
Cecilia has fixed this issue. I'm closing it.

closed 2022-02-28 by ajohlander