irfu/irfu-matlab #88: solo_local_file_db

sergiotr, 2022-03-08:
I am trying to load solo data using solo.db_init and solo.db_get_ts using a local file database without success.

solo.db_init('local_file_db','/Volumes/STOREX/data/solo/');
Tint = irf.tint('2020-08-28T06:00:00Z/2020-08-28T12:59:59Z');
Np = solo.db_get_ts('solo_L2_swa-pas-grnd-mom','N', Tint);

What is the folder hierarchy that needs to be present for this code to work?
So far I have tried:
/Volumes/STOREX/data/solo/L2/swa-pas-grnd-mom/2020/08/*.cdf
/Volumes/STOREX/data/solo/solo/L2/swa-pas-grnd-mom/2020/08/*.cdf

but none of these seem to work. Some logic has changed with respect how these commands used to work for mms?
I appreciate advice on how to build a local file database for solo and be able to use these very useful commands.
Many thanks in advance,
Sergio Toledo-Redondo

ErikPGJ, 2022-03-09:
I do not know the logic for MMS. I admit the logic for which directories will be used by solo_local_file_db is not at all obvious, but I think it is designed to handle what is on brain.irfu.se:/data/solo/ and therefore not entirely consistent, but almost. You may have triggered special behaviour when encountering certain subdirectories (wild guess). Are you using directories mounted from brain.irfu.se:/data/solo/?

Below commands work for me when using the exact paths on brain.irfu.se .

    solo.db_init('local_file_db', '/data/solo/data_irfu');     % Access to /data/solo/data_irfu/latest/rpw/
    solo.db_init('local_file_db', '/data/solo/');              % Access to /data/solo/remote/data/, /data/solo/soar/
    solo.db_init('local_file_db', '/data/solo/data_manual/');  % Access to /data/solo/data_manual/

where I have absolute paths like

    /data/solo/data_irfu/latest/rpw/L2/lfr_wf_e/2022/03/solo_L2_rpw-lfr-surv-swf-e_20220301_V01.cdf
    /data/solo/remote/data/L3/lfr_density/2020/12/solo_L3_rpw-bia-density-cdag_20201231_V02.cdf
    /data/solo/soar/swa/L2/swa-pas-grnd-mom/2021/11/solo_L2_swa-pas-grnd-mom_20211130_V02.cdf
    /data/solo/data_manual/mag/L2/mag-rtn-normal-1-minute/2021/01/solo_L2_mag-rtn-normal-1-minute_20210127_V01.cdf

The above example for data_manual/ is the most "general" case for accessing an arbitrary directory with non-RPW data (containing a standard directory structure) but it only works if there is no "soar" subdirectory, and only for non-RPW data.

Ultimately, the directory logic is implemented in method get_remotePrefix() in

    irfu-matlab_bicas/mission/solar_orbiter/solo_local_file_db.m .

It has four cases for how to interpret the specified directory path.
(a) Two cases for when looking for RPW data, and
(b) two cases for when looking for non-RPW data.
I have tried to clarify the source code comments on this somewhat (it is a short function). Look at it to understand what happens 

Beware that there is an overlap of datasets (on brain.spis.se) between
(1) /data/solo/data_irfu/latest/
("Bleeding edge" BIAS datasets produced internally at IRFU; Everything is labelled V01, even if it is not because datasets are inofficial anyway.)
and
(2) /data/solo/remote/data/
(all RPW datasets produced internally at LESIA)

sergiotr, 2022-03-10:
Dear Erik,
many thanks for the detailed reply. I am not mounting from brain.irfu.se, I am mounting my own database from a local server at my institution. I now understand how to build the folder hierarchy if I want to make it compatible with what solo.db_get_variable expects.
Many thanks for adding comments to the source code, I believe these will be helpful for other people as well not using the database at IRF.
I believe your reply solves the issue I raised. Best,
Sergio

ErikPGJ, 2022-03-10:
Good to hear, Sergio. I close the issue then.