irfu/irfu-matlab #93: SolO IRFU quicklooks crash: Value must be a 1x2 vector...

ErikPGJ, 2022-05-30:
This issue is not fully researched yet, but I want it documented anyway. It likely needs to be resolved by either @konradSteinvall or @ErikPGJ.

### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

This issue was found in 6cd2caf which is technically not the latest version, but the relevant code has not been updated since then.

### Step 2: Describe your environment
 * irfu-matlab branch (e.g. ```master```, ```devel```): devel
 * Matlab version used (e.g. ```Matlab R2020b```): R2020b
 * Operating system (e.g. ```Windows 10```, ```Mac OS X 10.15.6```, ```Linux Ubuntu 20.04```): Ubuntu 18.04.6 LTS (brain/spis)
 - [X] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem

#### Expected behavior

Generate Solar Orbiter IRFU quicklooks for the relevant time interval using @konradSteinvall 's code under +solo/quicklook_main.m.

#### Actual behavior

Code crashes. Two observed cases so far that seem to be caused by the same problem.
[spis: /home/erjo/logs/so_irfu_quicklooks.2022-05-19_11.29.51.log](https://github.com/irfu/irfu-matlab/files/8797927/so_irfu_quicklooks.2022-05-19_11.29.51.log)
[spis: /home/erjo/logs/so_irfu_quicklooks.2022-05-24_19.08.42.log](https://github.com/irfu/irfu-matlab/files/8797930/so_irfu_quicklooks.2022-05-24_19.08.42.log)
Ex: 
```
purging solo_L3_rpw-bia-efield-10-seconds_20210209_V01.cdf
purging solo_L3_rpw-bia-density-10-seconds_20210209_V01.cdf
purging solo_L2_mag-rtn-normal_20210209_V02.cdf
  [warning: get_variable(60)] dataobj:get_variable:cyclic_depCyclic dependecy: Epoch <-> Half_interval
  [warning: get_variable(60)] dataobj:get_variable:cyclic_depCyclic dependecy: Epoch <-> Half_interval
  [warning: get_variable(60)] dataobj:get_variable:cyclic_depCyclic dependecy: Epoch <-> Half_interval
  [warning: get_variable(60)] dataobj:get_variable:cyclic_depCyclic dependecy: Epoch <-> Half_interval
  [warning: get_variable(60)] dataobj:get_variable:cyclic_depCyclic dependecy: Epoch <-> Half_interval
  [warning: get_variable(60)] dataobj:get_variable:cyclic_depCyclic dependecy: Epoch <-> Half_interval
exception.message=Value must be a 1x2 vector of numeric type in which the second element is greater than the first element or is Inf.
quicklooks_24_6_2_h, row 44
quicklook_main, row 221
[?1l>STDERR: Command exited with non-zero status 99
```

#### Steps to reproduce:
Have not yet tried to reproduce, but the code crashed when executing below calls on brain/spis (see logs):
- `solo.quicklook_main('/home/erjo/so_irfu-matlab_qli//mission/solar_orbiter/+solo/irf_logo.png', '/data/solo/data_yuri/', '/data/solo/EJ_temp/qli//so_irfu_quicklooks_cron_2022-05-24_19.08.42', '1', '1', '2021-12-20T00:00:00', '2022-05-24T00:00:00');`
- `solo.quicklook_main('/home/erjo/so_irfu-matlab_qli//mission/solar_orbiter/+solo/irf_logo.png', '/data/solo/data_yuri/', '/data/solo/EJ_temp/qli//so_irfu_quicklooks_cron_2022-05-19_11.29.51', '1', '1', '2021-01-24T00:00:00', '2022-01-07T00:00:00');`

NOTE: Calls took 20h and 100h, but the duration should be unrelated to the bug.

ErikPGJ, 2022-05-30:
Judging from the completed plots, the error occurs when trying to plot data for:
- 2021-08-29
- 2022-01-12

thomas-nilsson-irfu, 2022-06-17:
I just came across this issue, the error message is typically generated by trying to set the X or Y limits to the same value, one quick example:

```matlab
fig = figure;
plot(1:2, 2:3, '*-');
ax = fig.Children;
ax.YLim % just output the automatic value
ax.YLim = [min(1), max(1)]; % try to set an incorrect range, result in the error message "Value must be a 1x2 vector of...."
```
It appears the code in question does have something similar (which could fail if for these dates `B` has only one value).
https://github.com/irfu/irfu-matlab/blob/c54ad3d8c876368bc43ca7a40f79b45978c2e2d9/mission/solar_orbiter/%2Bsolo/quicklooks_24_6_2_h.m#L44

---
Secondly, it appears the code in question is using hard coded values for fundamental constants, which should be avoided as much as possible.. We have a specific function for that called `irf_units` (see for instance e99c9c2a3b1d1f8d4cec30d783757f970a0ba839). Things like the following should be avoided whenever possible:
https://github.com/irfu/irfu-matlab/blob/c54ad3d8c876368bc43ca7a40f79b45978c2e2d9/mission/solar_orbiter/%2Bsolo/quicklooks_24_6_2_h.m#L17-L18
especially since it seems to suggest proton mass is measured in fractions of kilometres ;)

ErikPGJ, 2022-08-08:
Likely other sighting of same bug: Crashes for the call
`solo.quicklook_main('', '/data/solo/data_yuri/', '/home/erjo/temp/qli_cron/so_qli_cronhelper_cron_MTEST.2022-08-05_13.03.09/interm_output/so_qli_cronhelper_cron_2022-08-05_13.03.09', '1', '1', '2022-01-01T00:00:00', '2022-02-01T00:00:00');`
. 64.8 h wall time, but likely unrelated to bug.

Last completed plots for 2022-01-11 which implies that crash probably occurs for 2022-01-12 data (as seen before).

ErikPGJ, 2022-08-08:
`9c835bd6 JordiBoldu (2022-08-02 12:29:54 +0200) Updated quicklook plots` fixes 2022-01-12 so probably fixes bug.

ErikPGJ, 2022-08-10:
This appears solved by `7a6b2352 JordiBoldu (2022-08-08 13:41:25 +0200)`.

closed 2022-08-10 by ErikPGJ