irfu/irfu-matlab #108: Issue with using "Example_MMS_particle_DEFlux.m"

Krushna-100, 2022-10-21:
// TODO (you): please remove comments like this line and fill in this form as best you can, checking relevant boxes, etc.

### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

### Step 2: Describe your environment
 * irfu-matlab branch (e.g. ```master```, ```devel```): ____master
 * Matlab version used (e.g. ```Matlab R2020b```): ____R2018b
 * Operating system (e.g. ```Windows 10```, ```Mac OS X 10.15.6```, ```Linux Ubuntu 20.04```): ____Windows 10
 - [ ] A 64-bit installation of Matlab was used. yes

### Step 3: Describe the problem

While running "Example_MMS_particle_DEFlux.m", I am getting an error message for line no. 14. 

Dot indexing is not supported for variables of this type.

Error in mms.get_data/get_ts (line 946)
            theta = dist.DEPEND_2.data;

Error in mms.get_data (line 525)
        res = get_ts('skymap');

Error in c_eval (line 48)
    for cl_id=sc_list_1, evalin('caller', irf_ssub(ev_str, cl_id)), end

The line no. 13 which is for electron distribution is executed smoothly, whereas, line no. 14 which is for ion distribution giving an error message.

I will be pleased enough to get your early suggestions and help.

Thank you.

#### Expected behavior

#### Actual behavior

#### Steps to reproduce:
  // TODO (you): describe how to reproduce the problem. (Or if you prefer: you can fill in the "Relevant code" section below).
  1. _____
  2. _____
  3. _____

#### Relevant code:

  ```
    // TODO (you): insert code here to reproduce the problem.
    // (Or if you prefer: you can describe it in the "Steps to reproduce" section above).
  ```

thomas-nilsson-irfu, 2022-10-21:
Dear @Krushna-100,

may I ask you ti confirm that you have downloaded the corresponding `mms3_fpi_brst_l2_dis-dist` files and copied them to their corresponding folders (`mms3\fpi\brst\l2\dis-dist\2015\10\30\`) along with the `mms3_fpi_brst_l2_des-dist` files to their folders (`mms3\fpi\brst\l2\des-dist\2015\10\30\`)?

The error you experienced `Dot indexing is not supported for variables of this type.` can typically be seen if trying the corresponding line `theta = dist.DEPEND_2.data;` on an empty `dist` variable, meaning it failed to identify and load the files FPI ion distribution files.

With best regards,

Thomas Nilsson

Krushna-100, 2022-10-21:
> Dear @Krushna-100,
>
> may I ask you ti confirm that you have downloaded the corresponding `mms3_fpi_brst_l2_dis-dist` files and copied them to their corresponding folders [...]

Dear Nilsson,

Thank you very much for your quick reply. Although I added initially the "mms3_fpi_brst_l2_des-dist" file to the folder, however, the "mms3_fpi_brst_l2_dis-dist" file was missing.
After adding the latter file, it worked perfectly fine.

Thank you so much once again for your valuable suggestion.

With best regards,
Krushna

closed 2022-10-21 by thomas-nilsson-irfu