irfu/irfu-matlab #113: Issue with using "get_hiresdistributions"

Becky-wq, 2023-02-20:
// TODO (you): please remove comments like this line and fill in this form as best you can, checking relevant boxes, etc.

### Step 1: Latest code?
- [ ] This issue is present in the most recent code (i.e. after running ```git pull``` this is still a problem).

### Step 2: Describe your environment
 * irfu-matlab branch (e.g. ```master```, ```devel```): __master__
 * Matlab version used (e.g. ```Matlab R2020b```): __Matlab R2019b__
 * Operating system (e.g. ```Windows 10```, ```Mac OS X 10.15.6```, ```Linux Ubuntu 20.04```): __Windows 10__
 - [x] A 64-bit installation of Matlab was used.

### Step 3: Describe the problem

Dear IRF team,

I want to get high-resolution FPI data use the get_hiresdistributions function. The description of this function mentions that “Dist Must be in PDist format and include usec_offsets in ancillary data”. I try to construct the PDist skymap use make_pdist, but the obtained PDist data has ePDist.ancillary.usec_offsets=[]. How can i get ancillary.usec_offsets? 
It will be helpful if you give suggestions on it.

Thanks in Advance,
Becky

#### Relevant code:

  ```
db_info = datastore('mms_db');
file  = [db_info.local_file_db_root 'mms1\fpi\brst\l2\des-dist\2015\10\16\mms1_fpi_brst_l2_des-dist_20151016103254_v3.3.0.cdf'];
tmpDataObj = dataobj(file);
ePDist = mms.make_pdist(tmpDataObj);
Distpart = mms.get_hiresdistributions(ePDist);

  ```

danbgraham, 2023-02-20:
Hi Becky, 

For this function to work the des-dist cdf file must be v3.4.0 (one of the data products needed is not in the v3.3.0 files so they will not work). There have been a couple of updates to these routines so using the newest version of irfu-matlab on devel branch is also probably needed. 

Best regards,
Daniel

Becky-wq, 2023-02-21:
Dear Daniel,

Thank you very much for your quick reply. After using the v3.4.0 des-dist cdf file and the newest version of irfu-matlab on devel branch, the get_hiresdistributions function worked perfectly fine. I have another question, for there is only v3.3.0 des-dist cdf file available before July 1, 2021. For these periods, how can i increase the time resolution of FPI data?

Thank you so much once again for your valuable suggestion.

With best regards,
Becky

danbgraham, 2023-02-21:
Hi Becky, 

For intervals with only v3.3.0 data it is not possible to use this function. One needs to know the time offsets for each measurement in the distribution, which is only supplied in the v3.4.0 data. 

Best regards,
Daniel

Becky-wq, 2023-02-21:
Dear Daniel,

Thank you very much for your help. I have one more question, the particle moments data calculated using the following code differ significantly from the original burst mode data, I'm not sure where the problem is.  

code:
    db_info = datastore('mms_db');
    file = [db_info.local_file_db_root 'mms1\fpi\brst\l2\des-dist\2022\08\17\mms1_fpi_brst_l2_des-dist_20220817074933_v3.4.0.cdf'];
    tmpDataObj = dataobj(file);
    ePDist = mms.make_pdist(tmpDataObj);
    t = irf.tint('2022-08-17T07:49:33.000Z/2022-08-17T07:49:43.000Z');
    SCpot_t = mms.get_data('V_edp_brst_l2',t,1);
    ePDist_t = ePDist.tlim(t);
    Distpart = mms.get_hiresdistributions(ePDist_t,4);
    moments = mms.hiresmoments(Distpart,SCpot_t);
    ne = moments.n_psd;
    Ve = moments.V_psd;

Thank you so much once again for your valuable suggestion.

With best regards,
Becky

danbgraham, 2023-02-24:
Hi Becky,

The primary issue for this specific interval seems to be the low energy electrons (internal photoelectrons). One can obtain densities and velocities closer to the original by neglecting these low energy electrons. For example using 
moments = mms.hiresmoments(Distpart,SCpot_t,'energyrange',[30 4e4]);

For this interval the spacecraft are in the magnetotail where the density is quite low. As a result there are generally low counting statistics, which will make these higher resolution moments very uncertain. Rapid fluctuations in the density and velocity may simply result for these low number of particle counts and the large associated uncertainties in computing these moments.

Best regards,
Daniel

Becky-wq, 2023-02-26:
Dear Daniel,

Thank you so much for your great patience and huge help. 

With best regards,
Becky

closed 2023-02-26 by Becky-wq