irfu/irfu-matlab #119: How to check the results of "scpot2ne"?

renyon, 2023-05-29:
Whilst attempting to replicate the results of Figure 1 from the paper at https://doi.org/10.1029/2018JA025874 using the mms.scpot2ne function, I have found that my computation of the electron density estimated from the spacecraft potential seems to be inaccurate. I am reaching out in the hope that you might be able to provide some guidance or assistance. I have included my MATLAB code below :
%%
    mmsId = 1;
    Tint = irf.tint('2016-11-14T12:57:47.5Z/2016-11-14T12:57:48.8Z');
     
    %% EDP FPI
    ne = mms.get_data('Ne_fpi_brst_l2',Tint,mmsId);
    Te = mms.get_data('Te_gse_fpi_brst_l2',Tint,mmsId); 
    SCpot = mms.get_data('V_edp_brst_l2',Tint,mmsId);
     
    %% 
    [neSC,Iph0,Tph0,Iph1,Tph1] = mms.scpot2ne(SCpot,ne,Te );
    h = irf_plot({SCpot,ne,neSC} );
    ylabel(h(1),'V edp brst l2')
    ylabel(h(2),'Ne fpi brst l2')
    ylabel(h(3),'neSC')
%%
[image]

Best regards

danbgraham, 2023-05-29:
Hi Yong, 

For the function mms.scpot2ne to give meaningful results one needs to consider an extended period of time, where the density changes significantly such as at a magnetopause crossing. For Figure 1 of the paper the entire burst mode interval was used to estimate density from the spacecraft potential, e.g., Tint = irf.tint('2016-11-14T12:55:54.00Z/2016-11-14T12:58:44.00Z');.

Best Regards,
Daniel

renyon, 2023-05-29:
Dear Daniel,

Thank you very much for your insightful advice！！！！！！
I now understand that for the mms.scpot2ne function to yield meaningful results, it needs to consider an extended period of time where the density undergoes significant changes. I adjust my usage of the function to include a time interval similar to the one you provided from your paper.
[image]

I appreciate your guidance. Also, I was wondering if there are any other considerations or common pitfalls I should be aware of when using the mms.scpot2ne to study density fluctuation, dNe, due to Low hybrid wave at magnetopause (I know that you have made many research achievements in this field)？ Any additional advice you could offer would be greatly appreciated.
 
Best Regards,
Yong

renyon, 2023-05-30:
> Hi Yong,
>
> For the function mms.scpot2ne to give meaningful results one needs to consider an extended period of time, where the density changes significantly such as at a magnetopause crossing. For [...]

Dear Daniel,

Thank you very much for your insightful advice！！！！！！
I now understand that for the mms.scpot2ne function to yield meaningful results, it needs to consider an extended period of time where the density undergoes significant changes. I adjust  usage of the function to include a time interval similar to the one you provided from your paper.
[image]

I appreciate your guidance. Also, I was wondering if there are any other considerations or common pitfalls I should be aware of when using the mms.scpot2ne to study density fluctuation, dNe, due to Low hybrid wave at magnetopause (I know that you have made many research achievements in this field)？ (I know that you have made many research achievements in this field) Any additional advice you could offer would be greatly appreciated.

Best Regards,
Yong

danbgraham, 2023-06-01:
Hi Yong, 
I think the main issue with using the spacecraft potential to look at density perturbations is that the spacecraft potential is strongly affected by external electric fields. The calculated density perturbations will be least reliable when there are strong electric fields present (for example when lower hybrid waves are present). 

Best Regards,
Daniel

renyon, 2023-06-02:
> 
Dear Daniel,

Thank you very much for your feedback and suggestions. Your perspective on the use of spacecraft potential makes a lot of sense. Your professional insights have been immensely helpful, and I appreciate your willingness to share your knowledge and experience.

Best wishes,
Yong