irfu/irfu-matlab #145: Question regarding downloading Cluster data

ChaoGai72, 2024-06-12:
When I tried to run Example_1 in package, it reported wrongs saying that wrong happens in downloading URL with 'urlreadwrite_legacy'
I have double checked my account and password in datastore.  In addition, I tried my account on another computer, it also reports the same wrong. 
Also, I can download data with example codes showed on their website so I think it's not a problem about my network as it reported. I have no idea with how to tackle this problem. Thanks in advance!
[image]

thomas-nilsson-irfu, 2024-06-12:
Dear @ChaoGai72,

first a few questions:

- Are you using the latest version of `irfu-matlab` (what is the status if you run `git pull`?), and which branch are you on? (`master`, `devel`, or some other one?)

If you are running on our `master` branch, if you change to our latest development branch `devel` is the problem still there?
(That is to say, change branch with `$ git checkout devel`,  and make a pull to be sure its up to date `$ git pull`).

- Which version of Matlab are you using?
Some versions of Matlab does not have the new function `webread()` which is preferable over the old `urlread()`.

PajNath, 2024-09-17:
Hi, 
I got exactly the same problem @ChaoGai72, then I downloaded the caa_download from the del branch as well, and it worked for a few data. 
Now, whataver spacecraft I tried to download (SC1, SC3 or SC4) or the year I get or very strange symbols while downloading and it eventually nothing is downloaded, or it looks like it is downloaded I get a cdf file but there are all corrupted. 
I've tried to look by myself what could be wrong or about the path, but I cannot find anything obvious.
I have the feeling something is wrong and have changed in the caa_download or the path to access the data?
Could you please help?